AntonsMindstorms/PUPRemote #3: lookup for mode number

ste7anste7an, 2023-08-09:
https://github.com/antonvh/PUPRemote/blob/7cffd7331b237a5f96c6d25186238bdeab2b913b/LPF2/pupremote.py#L61

When a message from the hub is received by the pup, it needs to lookup the command that belongs to that mode. Therefore, we need an index of the commands indexes by mode.

Proposal:
- add a new list: `self.mode_command=[]` to the `__init__` function
- the mode number is the size of the lpf2.modes after adding the new mode.
```
self.lpup.modes.append( self.lpup.mode(mode_name, size) )
mode=len(self.lpf2.modes)
self.mode_command.append(mode_name)
```

ste7anste7an, 2023-08-20:
Implemented in new code

antonvh, 2025-01-20:
Can this be closed?

closed 2025-01-20 by ste7anste7an