AntonsMindstorms/PUPRemote #21: please clarify advantage of LMS-ESP32 over pybricks built-in support for xbox controller

m10d, 2024-06-05:
Interested in your ESP device, but wondering what it brings to the solution beyond the direct connection Pybricks can manage on it's own with the xbox controller?  Thank you.

antonvh, 2024-06-06:
For xbox series x/s the advantage is that the controller remains paired, you don't have to pair everytime you run a program.

For other controllers the advantage is that they work. LMS-ESP32 supports all bluetooth controllers. Pybricks supports only Xbos series x/s.

m10d, 2024-06-08:
Thanks for your reply. That makes sense, and it will justify the cost to let me use 8bitdo controllers i already have. Giving up the port is problematic - but i should be able to easily use the technic-compatible servos i have from an rpi project?

I am not finding an up to date answer on pybricks block-program integration. Is it trivial to use one motor/sensor daisy-chained to the LMS from pybricks on a technic hub? Multiple inputs/outputs on one LMS?

Last - i don't see any stl's online for a system or technic compatible case; are you aware of any?

Thanks!

m10d, 2024-06-08:
> I'd like to help but:
> 
>     * I don't know what you mean with 'technic-compatible servos i have from an rpi project'
> 
>     * I don't know what you mean with 'use one motor/sensor daisy-chained to the LMS from pybricks on a technic hub'
> 
> 
> There is no case, but the board has the exact dimensions and holes of a 5x7 technic frame.

Sorry for the ambiguity.  I was referring to the following:

- I have lego-format "normal" pwm servos & motors of the same kind in your image 
[image]
- but I use pybricks not lego-oem (I have the "mindstorms" which they try to lock out from spike in v3, and I also have many technic hubs which of course lego chooses to limit functionality on.  Pybricks has better quality of life, too); I only find old answers on if the `lms` / `uartremote` etc work with pybricks or with what limitations
    - is it possible to make this work with pybricks `blocks`, or just python code?  (specifically can my kids control motors & read from motor encoders & sensors on their own - I can write building block code as necessary, as long as things show up in blocks for them to use)
- I was hoping to find clarity on what the on-brick (eg 88012 or other lego hubs) looks like when multiple motors and/or sensors are connected to the `lms` - what sort of limitations are there, any significant responsiveness issues for hobby-level robots & RC models?

really appreciate the effort you put in to answer these, thank you!

antonvh, 2024-06-08:
Take a good look at this project: it has all the features you are looking for. In short: use pupremote, not uartremote.

https://www.antonsmindstorms.com/2024/04/06/robot-ball-kicker-project-building-a-lego-spike-prime-robot-with-openmv-vision/

antonvh, 2024-06-08:
Limitation for multiple motors is current. The hub only supplies 1.2A, enough for max 4 servos.

m10d, 2024-06-13:
> Limitation for multiple motors is current. The hub only supplies 1.2A, enough for max 4 servos.

In the product page it says "optional buck controller" for powering hobby servos.  I have these 4.8-6v servos https://www.aliexpress.us/item/3256803974452036.html similar to the product picture; obviously there is the overall current/power limit from the lego hub as you mention, but what is needed to run these directly from the LMS?

antonvh, 2024-06-15:
Nothing extra is needed. The buck is included and you can run 4 of them
from the lms.

m10d, 2024-06-15:
Perfect, ordered several!

closed 2025-01-20 by antonvh