AntonsMindstorms/PUPRemote #23: Strange behavior with Pybricks, LEGO remote control and servo

Werner-G, 2024-11-12:
I want to use the small Lego remote to control a Geekservo. I am using this Pybricks-code:
```
# Servo-Startwinkel und -Grenzen
servo_value = 0  # Startwert des Servos im Bereich -127 bis +127
servo_min_value = -127
servo_max_value = 127
angle_step = 10  # Schrittweite des Servo-Werts

# Funktion zur Umrechnung von Servo-Wert auf Winkel (0 bis 179 Grad)
def servo_value_to_angle(value):
    return (value + 128) * 179 // 256

# Steuerung des Roboters mit der LEGO-Fernbedienung
def control_robot_with_remote():
    global servo_value
    while True:
        pressed = remote.buttons.pressed()

        # Basissteuerung über den Servo (Button.LEFT und Button.RIGHT)
        if Button.LEFT in pressed:
            if servo_value > servo_min_value:
                servo_value -= angle_step
                bp.servo(1, servo_value_to_angle(servo_value))
            else:
                print("Servo hat die minimale Position erreicht.")
        
        elif Button.RIGHT in pressed:
            if servo_value < servo_max_value:
                servo_value += angle_step
                bp.servo(1, servo_value_to_angle(servo_value))
            else:
                print("Servo hat die maximale Position erreicht.")
```

The Servo is doing as described, but when initialing in the beginning and also when it comes to 0-position, the servo runs abrupt to a endpoint-position and continues. This might be a hardware-problem, but I don't know how to solve it.

antonvh, 2024-11-12:
This seems more like bluepad issue. I asked @ste7anste7an .

ste7anste7an, 2024-11-13:
Hi Werner,

I looked into your program. Actually there is only one small bug, and that
is that the value returned by your servo_value_to_angle function sometimes
becomes negative because this comparison allows for values to become
smaller than servo_min_value

 if servo_value > servo_min_value:
                servo_value -= angle_step

Resulting in a negative angle passed to the servo. Since the servo angle is
passed as an unsigned integer, this negative value becomes a large
positive value, making the servo move from around zero position to the
maximum  angle. You can replace that by:

if Button.LEFT in pressed:
            if servo_value > servo_min_value + angle_step:
                servo_value -= angle_step
                bp.servo(0, servo_value_to_angle(servo_value))

Let me know if this works for you,

As a remark, I would suggest that when you encounter a problem, to provide
us with the complete program, inclusing versions and hardware on which you
run it, instead of just a snippet. That makes reproducing your problem
easier.

kind regards,
Stefan

Werner-G, 2024-11-13:
Hi Stefan,

thanks for your response. Sorry, still the same behavior. Here is the complete code:
```
from pybricks.hubs import TechnicHub
from pybricks.pupdevices import Motor, Remote
from pybricks.parameters import Port, Button, Direction, Stop, Color
from pybricks.tools import wait
from bluepad import BluePad

# Initialisiere den Technic Hub
hub = TechnicHub()

# Initialisiere BluePad an Port A für den Servo (Basis)
bp = BluePad(Port.A)

# Initialisiere die LEGO-Fernbedienung
remote = Remote()

# Initialisiere die Schulter- und Ellbogenmotoren
shoulder_motor = Motor(Port.C, Direction.COUNTERCLOCKWISE, [12, 36])  # Schulter
elbow_motor = Motor(Port.D, Direction.COUNTERCLOCKWISE, [12, 36])  # Ellbogen

# Geschwindigkeit für die Motoren
speed = 200  # Geschwindigkeit in Grad/Sekunde

# Servo-Startwinkel und -Grenzen
servo_value = 0  # Startwert des Servos im Bereich -127 bis +127
servo_min_value = -127
servo_max_value = 127
angle_step = 10  # Schrittweite des Servo-Werts

# Funktion zur Umrechnung von Servo-Wert auf Winkel (0 bis 179 Grad)
def servo_value_to_angle(value):
    return (value + 128) * 179 // 256

# Steuerung des Roboters mit der LEGO-Fernbedienung
def control_robot_with_remote():
    global servo_value
    while True:
        pressed = remote.buttons.pressed()

        # Basissteuerung über den Servo (Button.LEFT und Button.RIGHT)
        if Button.LEFT in pressed:
            if servo_value > servo_min_value + angle_step:
                servo_value -= angle_step
                bp.servo(1, servo_value_to_angle(servo_value))
            else:
                print("Servo hat die minimale Position erreicht.")
        
        elif Button.RIGHT in pressed:
            if servo_value < servo_max_value + angle_step:
                servo_value += angle_step
                bp.servo(1, servo_value_to_angle(servo_value))
            else:
                print("Servo hat die maximale Position erreicht.")

        # Schultersteuerung über die A-Tasten
        if Button.LEFT_PLUS in pressed:
            shoulder_motor.run(speed)
        elif Button.LEFT_MINUS in pressed:
            shoulder_motor.run(-speed)
        else:
            shoulder_motor.stop()

        # Ellbogensteuerung über die B-Tasten
        if Button.RIGHT_PLUS in pressed:
            elbow_motor.run(speed)
        elif Button.RIGHT_MINUS in pressed:
            elbow_motor.run(-speed)
        else:
            elbow_motor.stop()

        # Aktuelle Positionen der Motoren und des Servos ausgeben
        print(f"Servo value (Base): {servo_value} (Servo angle: {servo_value_to_angle(servo_value)}°)")
        print(f"Shoulder motor angle: {shoulder_motor.angle()}°")
        print(f"Elbow motor angle: {elbow_motor.angle()}°")

        # Wartezeit für die Fernbedienungsabfrage
        wait(100)

# Hauptprogramm
try:
    control_robot_with_remote()
finally:
    # Beenden: Alle Motoren stoppen und Hub-LED auf grün setzen
    shoulder_motor.stop()
    elbow_motor.stop()
    hub.light.on(Color.GREEN)

```
 Pybricks (20240825) firmware on  LMS-ESP32 V1-board

Werner-G, 2024-11-13:
Oh oh oh, my mistake! I had to re-calibrate the position of the servo during power on. When setting to the 0-position, all movements are correct when using the remote. Thank you Stefan, your hint was excellent!

ste7anste7an, 2024-11-13:
Hi Werner,

good that it works now.  Internally in the bluepad code, I use the standard
Arduino servo library, which takes an input between 0 and 180 degrees.

Regards,
Stefan

ste7anste7an, 2024-11-13:
> What I want is an angle from -90 to +90°, not from 0 to 180°. So the
> converson formula is wrong.

Hi Werner, I belive my servo's (GEEK) only work from 0 to 180. But then you
can take 90 as you reference point? The angle you initially calculate goes
from 0 to 180.

antonvh, 2025-01-20:
Is this still an issue? I think we need to harmonize servo handling across all our libraries. However 0-180 or -90 - 90 is no bug, but a design choice. You just have to program around it.

closed 2025-01-20 by antonvh