AntonsMindstorms/PUPRemote #25: Problème avec la camera huskylens

ammarbendjama, 2025-01-19:
Bonjour,
Je viens d'essayer le programme avec une carte lms-esp32 v1 et une camera huskylens : 

[image]

Résultat : il affiche toujours (0, 0, 0) et les moteurs ne tournent pas ?

antonvh, 2025-01-20:
Can you check what the LMS-ESP32 sees, if you run the script through Thonny?
It looks like there is no data from the Huskylens. Did you 'teach' it to detect the line you want? Does it show an arrow on the screen?

ammarbendjama, 2025-01-20:
j'ai le résultat suivant en testant directement sue lms-esp32 : ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4892
ho 0 tail 12 room 4
load:0x40078000,len:14896
load:0x40080400,len:4
load:0x40080404,len:3372
entry 0x400805b0
LMS-ESP32 defaults loaded, with rx=18, tx=19
Checking heartbeat, but not connected. Initializing.

Successfully connected to hub with senor id 62
Huskylens connected is False
Checking heartbeat, but line is dead. Re-initializing.

Successfully connected to hub with senor id 62

ammarbendjama, 2025-01-20:
j'ai le résultat suivant sur LMS-ESP32 :

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4892
ho 0 tail 12 room 4
load:0x40078000,len:14896
load:0x40080400,len:4
load:0x40080404,len:3372
entry 0x400805b0
LMS-ESP32 defaults loaded, with rx=18, tx=19
Checking heartbeat, but not connected. Initializing.

Successfully connected to hub with senor id 62
Huskylens connected is False
Checking heartbeat, but line is dead. Re-initializing.

Successfully connected to hub with senor id 62

antonvh, 2025-01-21:
It clearly says the huskylens is not connected. Check baudrate, io pins, and i2c/uart configuration in the huskylens menu.