AntonsMindstorms/PUPRemote #27: error handling in `add_command`

kai-morich, 2025-02-16:
In [add_command](https://github.com/antonvh/PUPRemote/blob/9d31809dda54adf32f37413076f6c742d7aa43ae/src/pupremote.py#L222) you print the error condition, but then the program continues. 
My WRO team does not look at console output as this mainly consists of boot messages they do not understand. Their program sets the Neopixel to _red_ at the beginning and to _green_ before the while loop. I recommend to raise an exception, then we would have noticed that it doesn't reach _green_.

With https://github.com/antonvh/PUPRemote/pull/26 I proposed a related improval, but that focused on Spike side.

antonvh, 2025-02-16:
I improved the print message and then raise in f31c0979fb69d6c8e4e3b5e19776e4e75dcce9ce. Does this resolve the issue?

antonvh, 2025-02-22:
6978369e62e3aa8c9f71775dc18536bcb0938031 should fix this

closed 2025-02-22 by antonvh

kai-morich, 2025-02-22:
isn't the condition and error message one off?

Should it be something like

    Command name must not be longer than 5 with power=True

instead of 

    Command name must be shorter than 5 with power=True

?

antonvh, 2025-02-22:
You're right. Changed it in 2bea8043831a8f038ad61121328d9e75f30d0c70