AntonsMindstorms/PUPRemote #28: error handling in add_command() instead of call(...)

kai-morich, 2025-04-19:
My teams ran into various confusing error messages, so I did some tests:

- commands added in different order -> wrong command called, as they are called by number instead of name
- spike call() with max(input spec length, output spec length) different to anton max spec length -> spike raises: ValueError: Expected <max-input/output-num-bytes> values
- spike with higher command number than anton spec -> 'Invalid argument'

above can be detected earlier in add_command() by  name + length check and returning a reasonable error instead of getting an error later in call(...). I will provide a PR

- anton with more than 16 commands -> permanently fails with 'Failed to connect to hub' -> Pybricks LUMP_MAX_EXT_MODE 15 limit
- anton def ... returns a result where none was defined -> fails with 'AssertionError: Payload is empty'
- spike with command number >=8 -> 'OSError: [Errno 19] ENODEV:' -> maybe Pybricks LUMP_MAX_MODE 7 limit or bug in MSG_INFO_PLUS8 handling?

above should be fixed too

- request and response packets have same length and the shorter of both is filled up with '0' -> other spec mismatches are not detectable
- spike call() with more input then spec -> rest ignored
- spike call() with less input then spec -> filled up with '0'
- anton def ... returns less result than spec -> filled up with '0'
- anton def ... returns more result than spec -> rest ignored
- format string 'BH' is padded to length 4. can be disabled by using '<BH'

above ok

antonvh, 2025-04-20:
Yes, Pybricks has a bug in MSG_INFO_PLUS8. Probably best to limit the number of commands to 8 in add_command.

Can you tell me more about this? 
> anton def ... returns a result where none was defined -> fails with 'AssertionError: Payload is empty'

kai-morich, 2025-04-20:
spike:
```
from pybricks.parameters import Port
from pupremote_hub import PUPRemoteHub

rh = PUPRemoteHub(Port.F)
rh.add_command('a', '', 'B')
print('a',rh.call('a', 1))
````

and lms-esp32 code:
```
from pupremote import PUPRemoteSensor

def a(*args):
    print('a', args)
    return 2

rs = PUPRemoteSensor(power=True)
rs.add_command('a', '', 'B')

while True:
    rs.process()
```

results in:
```
Successfully connected to hub with sensor id 62
a (1,)
Traceback (most recent call last):
  File "main.py", line 1, in <module>
  File "main_cmd_test.py", line 64, in <module>
  File "pupremote.py", line 279, in process
  File "lpf2.py", line 264, in send_payload
  File "lpf2.py", line 236, in load_payload
AssertionError: Payload is empty
```

antonvh, 2025-04-22:
I suppose you want a clearer error message? Like 'No to_hub_format defined, while command {cmd} returned {data} to hub' ?

kai-morich, 2025-04-24:
cPython has strict number of parameter check in `struct.pack()`, whereas micropython is lazy. It fills up with 0 and ignores superfluous parameters.

I would prefer strict check as hunting ignored args or args filled with 0 is cumbersome.
With this helper function inspired by micropython `struct_pack_into_internal` implementation, the number of arguments could be calculated in `add_command` and checked later in `call`.

```python
def struct_count_args(s):
    '''count number of required args'''
    count = 0
    repeat = 0
    for c in s:
        if c.isdigit():
            repeat = repeat*10 + int(c)
        elif c == 'x': # pad byte
            pass
        elif c == 's': # string
            count += 1
        elif c.isalpha():
            if repeat == 0:
                count += 1
            else:
                count += repeat
                repeat = 0
    return count
```

kai-morich, 2025-04-26:
number of arguments can be calculated even shorter with built in functions:
```
fmt='bb'
len(struct.unpack(fmt,bytearray(struct.calcsize(fmt))))
```
but compared to the fixes I already provided I consider this less important

antonvh, 2025-04-27:
Fixed in db2e2285612c06024579b8cf6ded5f16f3cf9b8d. Can you test?

closed 2025-04-27 by antonvh

kai-morich, 2025-04-28:
- you have to treat `'repr'` differently
- the check after _decode_ has no effect as it uses the specified format.
- check in `call()` should also be added to pupremote_hub.py
- if the command has no return statement or returns `None`, the check is skipped due to `if result is not None:`
- the "Result of {} ..." assert messages should use '{}' to have the command name enclosed in single quotes like other assert messages already do.

kai-morich, 2025-04-30:
found 2 more corner cases, see comments on your latest change