AntonsMindstorms/PUPRemote #34: Use of PUPRemoteHub.call() in multitask

ggramlich, 2025-06-08:
We just got started with PUPRemote and LMS-ESP32 v2.
Thanks to your great work, we were able to control a servo motor connected to the ESP32 from the Spike Prime Hub and pybricks within a few hours.

We use this version: https://raw.githubusercontent.com/antonvh/PUPRemote/refs/heads/main/src/pupremote_hub.py
Then we tried to use `PUPRemoteHub.call()` inside a `multitask()` https://docs.pybricks.com/en/latest/tools/#multitasking
We wrapped the call in an async function. The result was

```
  File "pupremote_hub.py", line 239, in call
TypeError: object of type 'wait' has no len()

```

I basically ended up, writing a `call_async` method adding an `await` to `self.pup_device.write`, `self.pup_device.read` and `wait`.

```
    async def call_async(self, mode_name: str, *argv, wait_ms: int = 0):
        mode = self.modes[mode_name]
        cmd = self.commands[mode]
        # Encode & write
        if FROM_HUB_FORMAT in cmd:
            payload = self.encode(cmd[SIZE], cmd[FROM_HUB_FORMAT], *argv)
            data_out = tuple(payload + b'\x00'*(cmd[SIZE] - len(payload)))
            await self.pup_device.write(mode, self._int8_to_uint8(data_out))
            await wait(wait_ms)
        # read
        data_in = bytes(struct.pack('%db' % len(await self.pup_device.read(mode)), *(await self.pup_device.read(mode))))
        result = self.decode(cmd[TO_HUB_FORMAT], data_in)
        return result[0] if len(result) == 1 else result

```

I have not tried this with a call that should actually return something, so it might be that the read does not work correctly.
It seems to work well when multitasking one servo and the drivebase. Multitasking two servos causes an error on the `pup_device.write`.

```
  File "pupremote_hub.py", line 157, in call_async
OSError: [Errno 11] EAGAIN: Try again later
```

If i delay the second call_async in the multitask by about 60ms, I can move the two servos at the same time (with the little delay). But when I do that repeatedly, they start to stutter and the red light on the LMS-ESP32 flickers. I assume that it cannot provide enough power to both servos at the same time.

It would be OK for us to only multitask with one servo at a time.
I am just wondering, if this is the correct way to do it.

ggramlich, 2025-08-19:
@antonvh Sorry for not being responsive.

This is great. Thanks for your efforts.
I have not come around to testing it. Hopefully this weekend, I will get to it.

antonvh, 2026-01-03:
I removed the complete method. No need to do `await update_channel_async`, it is non blocking. Look at the example at the bottom of pupremote.py . Also please use the main branch from now. Fixed in dcb1802ed96ebbf496fd898d631baca0ee5b245d

closed 2026-01-03 by antonvh